sequenceplanner/r2r #6: Do you have a plan to implement action?

OTL, 2021-05-06:
Thank you for your great work!

In the ROS1 era, we were able to use pub/sub to use ROS1 action, so we didn't need action implement actually.

But action is natively implemented (not composition of the topics) in ROS2.
So, supporting action seems to be very important.
What I want to do is just send the goal to Nav2.

Do you have a plan to support action? I'm thinking of trying to implement it if you don't think it.
(i don't know if I can do it)

m-dahl, 2021-05-09:
Thank you!

What is supported by this package has evolved due to my needs, and so far I have not really needed action support. However, I don't think it should be too much work and it would be nice to have as then the bindings start being sort of complete. I will take an initial look at the action code in rcl.

OTL, 2021-05-10:
Thank you!

I dug a little about action.
I thought that if there were Fib.action, the ROS2 system produces Fib_Request.msg, Fib_Feedback.msg and Fib_Result.msg, like service/client system. But it actually does not generate msg files and creates only idl files. Then I stopped and created this issue to get your help!

m-dahl, 2021-05-10:
I just started the work here #9

closed 2021-07-24 by m-dahl