sequenceplanner/r2r #13: Spin in another thread

OTL, 2021-06-15:
Sorry for the feature request again!

I want to create a node (and pub/sub or something) and spin in another thread.
We have to call `spin_once()` of the node, but the node does not implement `Send`. So it's not possible.

Could you give me some advice to make it possible?

closed 2021-06-16 by m-dahl

m-dahl, 2021-06-16:
Hi,

There is no real reason the Node cannot be send. I have been meaning to look over the thread safety of the functions and protect the handles to the rcl stuff to it can also be Sync in the future but I have not yet done this. 5ab5ab5 enables your usecase. I also updated the example `subscriber_with_thread.rs` to spin in another thread. Let me know if you run into issues with this.

I am currently working on implementing an async api in #14, which is a lot more elegant and convenient than passing callback fucntions everywhere. Can I ask you for feedback regarding keeping the old API or breaking with a bump in version number in that pr? Thanks!

OTL, 2021-06-17:
Thanks!

I think it's nice and OK to break the API.

m-dahl, 2021-06-18:
Thanks I will probably go that route.

commit 5ab5ab5e (2021-06-16): impl Send for Node. Closes #13