sequenceplanner/r2r #16: The Parameter support does not seem to match what ROS2 calls parameters

mschuckmann, 2021-07-27:
Looking at the parameters example code it appears that that this crates notion of parameters does not patch ROS2 notion of parameters. 

From my understanding of ROS2 parameters they are are attributes of a Node that can be get and set by other nodes, I'm new to this library (and to ROS2) but I don't really see how this library implements that functionality? It seems that it is just reading the values specified via `--ros-args` and adding them to a hash map for the code to use, is that correct?

m-dahl, 2021-07-28:
Hi,

I think you are right. From my understanding, parameters can be set by commandline, yaml files and via services available to other nodes. The services associated with updating parameters have not been implemented. I have updated the readme to clarify that this functionality is missing.

mschuckmann, 2021-07-28:
Thanks for confirming.

mschuckmann, 2021-07-28:
So any idea what it would take to implement parameters properly? I'd be willing to do the work but I'd probably need some hand holding.

m-dahl, 2021-07-29:
Hi,

I added simple updating in #17. Closing this issue in favor of that PR.

If you want to you could take a stab at implementing the services in that PR using the one I added already as a template. Depending on your usecase we can also merge to master early when you have what you need.

closed 2021-07-29 by m-dahl

mschuckmann, 2021-07-29:
Awesome thank you, I will take a look at the PR. At a minimum I'll need to
be able to set and get parameters and have the node be notified when a
parameter changes so that it can take action.