sequenceplanner/r2r #18: Example for a custom service

mschuckmann, 2021-07-29:
Sorry if this is something that should be obvious to an experienced ROS developer but how would one go about creating a custom service, the existing example uses a predefined service from ROS examples 

The readme discusses  auto generation of bindings and msg but it's unclear to me what the process is.

mschuckmann, 2021-07-29:
I think I figured this out, I just need to go through the steps of creating a package with a custom service using the ros2 tools, when I sourced the workspace environment my service messages were created and then so long as I ran `cargo build` from that environment the service types are available.  This tutorial was very helpful
https://docs.ros.org/en/foxy/Tutorials/Custom-ROS2-Interfaces.html

closed 2021-07-29 by mschuckmann

m-dahl, 2021-07-29:
Yes you got it. Additionally your can take a look at this repo for an example <https://github.com/m-dahl/r2r_minimal_node/> if you want to use colcon also for the rust stuff.