sequenceplanner/r2r #26: Is interprocess communication supported?

szandara, 2022-01-01:
One exciting feature of ROS2 is efficient inter-process communication to avoid serialization and network overhead for certain message exchanges. 

https://docs.ros.org/en/foxy/Tutorials/Intra-Process-Communication.html

Is it supported by r2r? I was not able to find any reference to it.
If not is there any plan to support it?

m-dahl, 2022-01-10:
Hi,

It is not currently supported. I am not familiar with how it's implemented, but if it is just an option when creating the node it should be straight forward to add support for it. (I.e. if its build into rcl rather than rclcpp).

kilpkonn, 2022-01-28:
I did some digging into this and it turns out the implementation for interprocess communication is in `rclcpp`, not `rcl` meaning it can't be easily supported in rust. However there is a motivation to move it to `rcl` or `rmw` meaning it might be possible/a lot easier in the future. More info here: https://github.com/ros2/design/issues/251