sequenceplanner/r2r #31: WrappedTypesupport private

aboseley, 2022-11-15:
I got stuck today trying to write a generic subscribe function for a set of r2r ROS types, because the `WrappedTypesupport` trait is private. All the ROS types I was using implemented this trait because they were auto-generated messages type.  So this feels like something I should be able to do.

kristoferB, 2022-12-21:
Good suggestion. When do you plan to implement this trait manually?

lopsided98, 2022-12-30:
I'm running into the same problem with `WrappedServiceTypeSupport`. I'm not implementing the trait manually; I would like to write a function that takes a generic service `Client` as a parameter, like this:
```rust
fn some_function<T: r2r::msg_types::WrappedServiceTypeSupport>(client: r2r::Client<T>) {
}
```

m-dahl, 2022-12-30:
I agree that it would be useful. I will take a look at this next week. Should be a simple fix.

closed 2023-01-17 by m-dahl

m-dahl, 2023-01-17:
Hi. These should now be accessible in 0.6.5.

commit af452ec6 (2023-01-17): Make WrappedTypesupport traits public. Closes #31