sequenceplanner/r2r #37: Build Fails ( ROS not sourced?)

SimeonOA, 2023-01-25:
I am building a DockerFile for a project that depends on r2r.

I am using ros2 and have made sure to source my ros2 installation before building my application.

I keep having the following error during `cargo build`:

```
error: failed to run custom build command for `r2r_rcl v0.3.4`

Caused by:
  process didn't exit successfully: `/gdp-router/target/debug/build/r2r_rcl-70fc35e866c7fffb/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=AMENT_PREFIX_PATH
  cargo:rerun-if-env-changed=CMAKE_INCLUDE_DIRS
  cargo:rerun-if-env-changed=CMAKE_LIBRARIES
  cargo:rerun-if-env-changed=CMAKE_IDL_PACKAGES
  cargo:rerun-if-env-changed=IDL_PACKAGE_FILTER

  --- stderr
  Generate bindings file '.../target/debug/build/r2r_rcl-33a8866d536db875/out/fe5011cc8ab0e2af6e01c3a254e65b60897d7140b7771f907c1a5a99bb6a987a/rcl_bindings.rs'
  thread 'main' panicked at 'Source your ROS!: NotPresent', /root/.cargo/registry/src/github.com-1ecc6299db9ec823/r2r_common-0.3.4/src/lib.rs:60:64
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
warning: build failed, waiting for other jobs to finish...
```

I have tried both rolling and humble and keep having this error.

Is there something I am missing or should be doing?

Thank you

SimeonOA, 2023-01-25:
Here is the error when I run with `'RUST_BACKTRACE=1'`:

```
error: failed to run custom build command for `r2r_rcl v0.3.4`

Caused by:
  process didn't exit successfully: `/gdp-router/target/debug/build/r2r_rcl-70fc35e866c7fffb/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=AMENT_PREFIX_PATH
  cargo:rerun-if-env-changed=CMAKE_INCLUDE_DIRS
  cargo:rerun-if-env-changed=CMAKE_LIBRARIES
  cargo:rerun-if-env-changed=CMAKE_IDL_PACKAGES
  cargo:rerun-if-env-changed=IDL_PACKAGE_FILTER

  --- stderr
  Generate bindings file '/gdp-router/target/debug/build/r2r_rcl-33a8866d536db875/out/fe5011cc8ab0e2af6e01c3a254e65b60897d7140b7771f907c1a5a99bb6a987a/rcl_bindings.rs'
  thread 'main' panicked at 'Source your ROS!: NotPresent', /root/.cargo/registry/src/github.com-1ecc6299db9ec823/r2r_common-0.3.4/src/lib.rs:60:64
  stack backtrace:
     0: rust_begin_unwind
               at /rustc/90743e7298aca107ddaa0c202a4d3604e29bfeb6/library/std/src/panicking.rs:575:5
     1: core::panicking::panic_fmt
               at /rustc/90743e7298aca107ddaa0c202a4d3604e29bfeb6/library/core/src/panicking.rs:65:14
     2: core::result::unwrap_failed
               at /rustc/90743e7298aca107ddaa0c202a4d3604e29bfeb6/library/core/src/result.rs:1791:5
     3: core::result::Result<T,E>::expect
               at /rustc/90743e7298aca107ddaa0c202a4d3604e29bfeb6/library/core/src/result.rs:1070:23
     4: r2r_common::setup_bindgen_builder
               at /root/.cargo/registry/src/github.com-1ecc6299db9ec823/r2r_common-0.3.4/src/lib.rs:60:32
     5: build_script_build::gen_bindings
               at ./build.rs:79:20
     6: build_script_build::run_bindgen
               at ./build.rs:31:13
     7: build_script_build::main
               at ./build.rs:9:5
     8: core::ops::function::FnOnce::call_once
               at /rustc/90743e7298aca107ddaa0c202a4d3604e29bfeb6/library/core/src/ops/function.rs:251:5
  note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
warning: build failed, waiting for other jobs to finish...
```

m-dahl, 2023-05-22:
Closing this due to inactivity. Please reopen if your issue remains.

closed 2023-05-22 by m-dahl