sequenceplanner/r2r #38: Build fails: unresolved import `RCUTILS_LOG_SEVERITY`

aresyy, 2023-02-02:
Hi there, I've tried to run the standard publish example, but got some issues.

What I did :

```
# clone the git repo
source /opt/ros/foxy/setup.bash
cd r2r
cargo build
```

Cargo error:

```
   Compiling r2r v0.6.6 (/app/r2r/r2r)
error[E0432]: unresolved import `RCUTILS_LOG_SEVERITY`
  --> r2r/src/utils.rs:63:13
   |
63 |         use RCUTILS_LOG_SEVERITY::*;
   |             ^^^^^^^^^^^^^^^^^^^^ use of undeclared type `RCUTILS_LOG_SEVERITY`

error[E0412]: cannot find type `RCUTILS_LOG_SEVERITY` in this scope
  --> r2r/src/utils.rs:62:28
   |
62 |     fn to_native(&self) -> RCUTILS_LOG_SEVERITY {
   |                            ^^^^^^^^^^^^^^^^^^^^ not found in this scope

error[E0425]: cannot find value `RCUTILS_LOG_SEVERITY_UNSET` in this scope
  --> r2r/src/utils.rs:65:35
   |
65 |             LogSeverity::Unset => RCUTILS_LOG_SEVERITY_UNSET,
   |                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^ not found in this scope

error[E0425]: cannot find value `RCUTILS_LOG_SEVERITY_DEBUG` in this scope
  --> r2r/src/utils.rs:66:35
   |
66 |             LogSeverity::Debug => RCUTILS_LOG_SEVERITY_DEBUG,
   |                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^ not found in this scope

error[E0425]: cannot find value `RCUTILS_LOG_SEVERITY_INFO` in this scope
  --> r2r/src/utils.rs:67:34
   |
67 |             LogSeverity::Info => RCUTILS_LOG_SEVERITY_INFO,
   |                                  ^^^^^^^^^^^^^^^^^^^^^^^^^ not found in this scope

error[E0425]: cannot find value `RCUTILS_LOG_SEVERITY_WARN` in this scope
  --> r2r/src/utils.rs:68:34
   |
68 |             LogSeverity::Warn => RCUTILS_LOG_SEVERITY_WARN,
   |                                  ^^^^^^^^^^^^^^^^^^^^^^^^^ not found in this scope

error[E0425]: cannot find value `RCUTILS_LOG_SEVERITY_ERROR` in this scope
  --> r2r/src/utils.rs:69:35
   |
69 |             LogSeverity::Error => RCUTILS_LOG_SEVERITY_ERROR,
   |                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^ not found in this scope

error[E0425]: cannot find value `RCUTILS_LOG_SEVERITY_FATAL` in this scope
  --> r2r/src/utils.rs:70:35
   |
70 |             LogSeverity::Fatal => RCUTILS_LOG_SEVERITY_FATAL,
   |                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^ not found in this scope
```

What I've tried:

I noticed it this might be something related to `r2r_rcl/build.rs` to generate `rcl_binding.rs`, here are r2r_rcl build outputs:

Output:

```
cargo:rerun-if-env-changed=AMENT_PREFIX_PATH
cargo:rerun-if-env-changed=CMAKE_INCLUDE_DIRS
cargo:rerun-if-env-changed=CMAKE_LIBRARIES
cargo:rerun-if-env-changed=CMAKE_IDL_PACKAGES
cargo:rerun-if-env-changed=IDL_PACKAGE_FILTER
cargo:rustc-link-search=native=/opt/ros/foxy/lib
cargo:rustc-link-lib=dylib=rcl
cargo:rustc-link-lib=dylib=rcl_logging_spdlog
cargo:rustc-link-lib=dylib=rcl_yaml_param_parser
cargo:rustc-link-lib=dylib=rcutils
cargo:rustc-link-lib=dylib=rmw
cargo:rustc-link-lib=dylib=rosidl_typesupport_c
cargo:rustc-link-lib=dylib=rosidl_runtime_c
```

stderr:

```
Generate bindings file '/app/r2r/target/debug/build/r2r_rcl-767eccf1419ba33f/out/5c2228f77f6ab7d9477fa2290f42d66457149b31bbc7a1dddb8b55a7bdfae55e/rcl_bindings.rs'
```

Seems like binding is generated. I compare `5c2228f77f6ab7d9477fa2290f42d66457149b31bbc7a1dddb8b55a7bdfae55e/rcl_bindings.rs` with `r2r/r2r_rcl/rcl_bindings.rs`, the previous one doesn't contain `enum RCUTILS_LOG_SEVERITY`, I think that cause this issue.

No idea where I should go after this. Any suggestions will be appreciated.

aresyy, 2023-02-02:
My environment:

MacOS Monterey 12.6 M1 pro chips
Example running in docker container:
ubuntu 20.04 
kernel 5.10.124-linuxkit 
aarch64

rustc 1.67.0 (fc594f156 2023-01-24)
cargo 1.67.0 (8ecd4f20a 2023-01-10)

m-dahl, 2023-02-02:
Hi,

Thanks. My initial guess is that we accidentally dropped support for foxy (CI for foxy was removed a while back, so I wouldn't have noticed). If foxy important to you? This can probably be fixed, but I am not sure it is worth it since foxy is quite old by now.

aresyy, 2023-02-02:
Thanks for responding.

You're right foxy will end of life on May 2023. Personally, I don't think it's worth fixing that.

For my team, we have some hardware that is hard to install ubuntu 22.04. Also, many modules are built with legacy C++ libraries, so it's hard to switch to a new GCC version and get direct support with humble. Those are my problems, I will go from here to see if I can do something. 

Thanks for explanation : )

aresyy, 2023-02-15:
@m-dahl 
Got some time today, I tested on my local
if I add `.allowlist_type("RCUTILS_.*")` on https://github.com/sequenceplanner/r2r/blob/master/r2r_rcl/build.rs#L80, then on ROS foxy it compile success. For my directly feeling, rust-bindgen shouldn't generate enum `RCUTILS_LOG_SEVERITY_WARN` for humble as well, no idea why it works.

m-dahl, 2023-02-17:
Great! I didn't look into why it works on humble but I made a PR with your fix. If CI is ok for galactic and humble I will merge it.

aresyy, 2023-02-20:
All I can think is galactic and humble's allowlist exports references `RCUTILS_LOG_SEVERITY` type. I can see CI pass, feel free to close this ticket.

m-dahl, 2023-02-20:
I pushed 0.6.7 with this fix. Thanks again.

closed 2023-02-20 by m-dahl