sequenceplanner/r2r #49: Parsing Rust file based on linebreak is not stable

JiangengDong, 2023-05-25:
```Rust
fn add_constants(key: &str, bindings: &str, constants: &mut HashMap<String, Vec<(String, String)>>) {
    let mut lines = bindings.lines();
    ...
}
```

In the PR for constants, the implementation is as above. However, this is not reliable and stable, because the result of `bindgen` does not guarantee to be of the correct format. 

When built in an environment that has `rustfmt` installed, the `msg_bindings.rs` is generated in the correct format. However, when built in an environment without `rustmt`, the `msg_bindings.rs` is generated as a single-line file, which in turn fails the whole building because the action-related constants are not generated correctly. 

Should we use `syn` to parse a rust file instead?

m-dahl, 2023-05-28:
Yes, using syn to parse it sounds much better, atleast as it is currrently. It would be even better if we can avoid parsing the bindgen-generated code in the future, it feels a bit like a hack to do it.

m-dahl, 2023-06-26:
Should be fixed as of #58