sequenceplanner/r2r #53: Compile only necessary packages to speed up compilation

jerry73204, 2023-06-07:
Currently, r2r compiles all discovered message types, which leads to a lengthy compilation time.
The idea is to let r2r acknowledge required ROS packages in `package.xml` (or `Cargo.toml`), and compile necessary packages.
The tricky part is to allow `r2r` to learn requested packages from dependent crates. It is usually done by feature flags or environment variables. A relevant discussion can be found here rust-lang/cargo#4121.

m-dahl, 2023-06-07:
Hi.

We have this though I realize it is not well documented. You can see a note about it under "building with cargo" [here](https://github.com/m-dahl/r2r_minimal_node#building-using-only-cargo). 

It is just a crude filter so it does not know about dependencies, so if you have nested message types you need to explicitly include all of the packages you use.

If you use the colcon way of building it will also only build the requested packages. I hope this helps.

jerry73204, 2023-06-07:
How does r2r discover requested packages when using colcon? Just curious.

jerry73204, 2023-06-09:
Thanks for your explanation. This feature should be well documented in README. It helps reducing lots of compilation time.

closed 2023-06-09 by jerry73204

m-dahl, 2023-06-20:
Added to readme as per your suggestion, thanks.