sequenceplanner/r2r #57: Support for dealine on subscribers

rdelfin, 2023-06-20:
I'm currently working on a ROS node that requires use of the [subscriber node feature](https://design.ros2.org/articles/qos_deadline_liveliness_lifespan.html). While I can see there's a method for [setting the deadline](https://docs.rs/r2r/latest/r2r/qos/struct.QosProfile.html#method.deadline), I can't find a way of setting up a callback, similar to the [deadline_callback](https://docs.ros2.org/galactic/api/rclcpp/structrclcpp_1_1SubscriptionEventCallbacks.html#a373f0ff83122110b3c1f8dc153cce5d1) option you can set in rclcpp. Is there an equivalent, or plan to add support for this feature?

rdelfin, 2023-06-20:
If there is no plan I'm looking into making a PR for this. Doesn't look _too_ complicated, but I'm sure I'll come to regret saying that

m-dahl, 2023-06-20:
You are right, there is no mechanism like this implemented, and I have no plans to add it anytime soon. Feel free to make a PR, I will support best I can. If the functionality is exposed in rcl it shouldn't be too much work I think.