sequenceplanner/r2r #62: Implementing Actions

OmnipotentEntity, 2023-07-31:
I was wanting to implement a custom action server; however, I cannot find any documentation or examples that address this.  I was able to find the [action server](https://github.com/sequenceplanner/r2r/blob/cc84beb17d19b83b8d84f5ad7910021070c4dd2a/r2r/examples/action_server.rs) example.  However, this example seems to merely be a frontend for the `Fibonacci` example action, which I was unable to find the source for.

I get the feeling that I'm almost certainly missing something easy, but I couldn't find any other users asking about this on this issues page or elsewhere (such as the ROS discourse forum).

m-dahl, 2023-08-01:
Hi. 

Basically you want to create your own action message definitions. Then you can implement your action server in the same way as in the action server example, but using YourMessage::Action instead of Fibonacci::Action.

If you take a look at the [minimal example repo](https://github.com/m-dahl/r2r_minimal_node) you can see how it defines a custom service type and implements the server logic for it. Creating an action will be very similar (in fact that repo contains an example action message file already, but no action server is implemented for it.)

I hope this helps to get started. Let me know if you get stuck.

OmnipotentEntity, 2023-08-04:
Maybe I'm misunderstanding how this is supposed to work, but I'm attempting to add in the following code immediately before the spin loop:

```
    let action_server = node.create_action_server::<SayHello::Action>("/say_hello")?;
    tokio::task::spawn(async move {
        action_server.for_each(|req| {
            println!("request: {}", req.goal.to_say);
            if req.goal.to_say == "Hello" {
                let _ = req.accept().expect("could not send response");
            } else {
                let _ = req.reject().expect("could not send response");
            }
            future::ready(())
        })
        .await
    });
```

This is compiling, but only seems to be 1/3 of an action server.  How do I (for instance) publish feedback on the action server topic or handle results requests through the results service?

m-dahl, 2023-08-10:
Hi, sorry I forgot to reply. Check [these lines](https://github.com/sequenceplanner/r2r/blob/cc84beb17d19b83b8d84f5ad7910021070c4dd2a/r2r/examples/action_server.rs#L53-L76) of the action server example. Now you throw away the goal and cancel tuple on `accept()`, but if you write it as the example you can publish feedback with `g.publish_feedback()` and succeed or abort etc. `cancel` is a stream of cancel requests if you want to handle goal cancelling.

closed 2023-08-15 by OmnipotentEntity