sequenceplanner/r2r #69: Regression: Commit e33a57c causes "'rcl/rcl.h' file not found"

jerry73204, 2023-11-22:
The commit e33a57c drops include search paths such as `rcl/rcl.h`, causing bindgen to fail. The commit discards the paths from `CMAKE_INCLUDE_DIRS`, which is set by `source install/setup.*sh`. This change must be reverted.

```
    Updating crates.io index
   Compiling r2r_rcl v0.8.0
   Compiling r2r_actions v0.8.0
error: failed to run custom build command for `r2r_actions v0.8.0`
note: To improve backtraces for build dependencies, set the CARGO_PROFILE_DEV_BUILD_OVERRIDE_DEBUG=true environment variable to enable debug information generation.

Caused by:
  process didn't exit successfully: `/my-project/src/artc_obj_detect/target/debug/build/r2r_actions-23cc6fc28ac64f3c/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=AMENT_PREFIX_PATH
  cargo:rerun-if-env-changed=CMAKE_PREFIX_PATH
  cargo:rerun-if-env-changed=CMAKE_IDL_PACKAGES
  cargo:rerun-if-env-changed=IDL_PACKAGE_FILTER
  cargo:rerun-if-env-changed=ROS_DISTRO
  cargo:rustc-cfg=r2r__ros__distro__humble

  --- stderr
  Generate bindings file '/my-project/src/artc_obj_detect/target/debug/build/r2r_actions-0e340de0d2d46a2f/out/e8cb6e9dff909737e080a658a09734b2760d27887417748b639d0bdf6ec58362/action_bindings.rs'
  src/action_wrapper.h:1:10: fatal error: 'rcl_action/rcl_action.h' file not found
  thread 'main' panicked at /home/aeon/.cargo/registry/src/index.crates.io-6f17d22bba15001f/r2r_actions-0.8.0/build.rs:94:10:
  Unable to generate bindings: ClangDiagnostic("src/action_wrapper.h:1:10: fatal error: 'rcl_action/rcl_action.h' file not found\n")
  stack backtrace:
     0: rust_begin_unwind
               at /rustc/79e9716c980570bfd1f666e3b16ac583f0168962/library/std/src/panicking.rs:597:5
     1: core::panicking::panic_fmt
               at /rustc/79e9716c980570bfd1f666e3b16ac583f0168962/library/core/src/panicking.rs:72:14
     2: core::result::unwrap_failed
               at /rustc/79e9716c980570bfd1f666e3b16ac583f0168962/library/core/src/result.rs:1652:5
     3: core::result::Result<T,E>::expect
     4: build_script_build::generate_bindings
     5: build_script_build::run_bindgen
     6: build_script_build::main
     7: core::ops::function::FnOnce::call_once
  note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
warning: build failed, waiting for other jobs to finish...
error: failed to run custom build command for `r2r_rcl v0.8.0`
note: To improve backtraces for build dependencies, set the CARGO_PROFILE_DEV_BUILD_OVERRIDE_DEBUG=true environment variable to enable debug information generation.

Caused by:
  process didn't exit successfully: `/my-project/src/artc_obj_detect/target/debug/build/r2r_rcl-ecf2c203705bd96f/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=AMENT_PREFIX_PATH
  cargo:rerun-if-env-changed=CMAKE_PREFIX_PATH
  cargo:rerun-if-env-changed=CMAKE_IDL_PACKAGES
  cargo:rerun-if-env-changed=IDL_PACKAGE_FILTER
  cargo:rerun-if-env-changed=ROS_DISTRO
  cargo:rustc-cfg=r2r__ros__distro__humble

  --- stderr
  Generate bindings file '/my-project/src/artc_obj_detect/target/debug/build/r2r_rcl-7e07310182565eb6/out/e8cb6e9dff909737e080a658a09734b2760d27887417748b639d0bdf6ec58362/rcl_bindings.rs'
  src/rcl_wrapper.h:9:10: fatal error: 'rcl/rcl.h' file not found
  thread 'main' panicked at /home/aeon/.cargo/registry/src/index.crates.io-6f17d22bba15001f/r2r_rcl-0.8.0/build.rs:99:10:
  Unable to generate bindings: ClangDiagnostic("src/rcl_wrapper.h:9:10: fatal error: 'rcl/rcl.h' file not found\n")
  stack backtrace:
     0: rust_begin_unwind
               at /rustc/79e9716c980570bfd1f666e3b16ac583f0168962/library/std/src/panicking.rs:597:5
     1: core::panicking::panic_fmt
               at /rustc/79e9716c980570bfd1f666e3b16ac583f0168962/library/core/src/panicking.rs:72:14
     2: core::result::unwrap_failed
               at /rustc/79e9716c980570bfd1f666e3b16ac583f0168962/library/core/src/result.rs:1652:5
     3: core::result::Result<T,E>::expect
     4: build_script_build::gen_bindings
     5: build_script_build::run_bindgen
     6: build_script_build::main
     7: core::ops::function::FnOnce::call_once
  note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
```

m-dahl, 2023-11-30:
Hi,

Thanks. I hastily released 0.8.1 which adds back the cmake include dirs. Can you try it? (You need to grab a new r2r_cargo.cmake)

I will (when I get the time) add a CI action that builds the minimal node with colcon soon, I realize we need that.

m-dahl, 2023-12-05:
Hmm. I am not sure what is up here. `/opt/ros/humble' exists in your `AMENT_PREFIX_PATH` so it should have been picked up. It works in the CI action which uses ubuntu + humble. What platform is this on? Are you building with cargo or colcon?

alecrivers, 2023-12-09:
I just stumbled upon an issue which may be the same thing. Compiling on Linux, this line: https://github.com/sequenceplanner/r2r/blob/50b5f9da27cba2659fc30c281f2198832444ae74/r2r_common/src/lib.rs#L69 adds a semicolon rather than a colon. In my case, that meant that the ament string became:

```home/alec/830/build/ros2_ws/install/realsense2_description:/home/alec/830/build/ros2_ws/install/realsense2_camera:/home/alec/830/build/ros2_ws/install/realsense2_camera_msgs:/opt/ros/humble;/home/alec/830/build/ros2_ws/install/realsense2_description:/home/alec/830/build/ros2_ws/install/realsense2_camera:/home/alec/830/build/ros2_ws/install/realsense2_camera_msgs```

That means that `/opt/ros/humble` gets joined with the next directory and is basically broken. Hence the include woes. I was able to fix it by moving up the definition of `split_char` and replacing that line with `ament_str.push(&split_char.to_string());`. (Apologies for not submitting a PR; I am new to contributing to open source stuff and don't know the right process.)

m-dahl, 2023-12-09:
Thanks, good catch! If you have an interest in learning how to do it, I encourage you to try and make a PR (don't worry about the process or messing something up - you can't really break anything). If not I will fix it in the coming days.

alecrivers, 2023-12-10:
OK, I created a PR.

PS, thanks for a cool project! I'm writing software for a robot and challenging myself by doing it in Rust (I'm a C++ developer) and r2r is very ergonomic even for a Rust and ROS2 newbie like myself.

jerry73204, 2023-12-10:
@alecrivers Your patch fixes my issue. You save my day. Thanks.

m-dahl, 2023-12-10:
@alecrivers Thanks for the PR and the kind words. r2r was my "learn rust" project initially aswell 😄 

@jerry73204 Thanks for verifying that it works.

I will make a patch release shortly.

closed 2023-12-10 by m-dahl

m-dahl, 2023-12-11:
FYI I have now released 0.8.2 with the fix included.