sequenceplanner/r2r #79: How to create a new r2r::builtin_interfaces::msg::Time to represent the current timestamp? I can't seem to find fn

xiaocijun, 2023-12-26:
use image::{GenericImageView, ImageBuffer, RgbImage};
use r2r::*;
use core::time;
use std::time::{Duration};
use std::time::{SystemTime, UNIX_EPOCH};

fn main() {
    // Initialize the R2R context and create a node
    let ctx = Context::create().unwrap();
    let mut node = Node::create(ctx, "rust_image_publisher", "").unwrap();

    // Create a publisher for the image topic
    let publisher = node
        .create_publisher::<sensor_msgs::msg::Image>("camera/image_raw", QosProfile::default())
        .unwrap();

    // Set the loop rate (e.g., 10 Hz)
    let loop_rate = Duration::from_millis(100);

    loop {
        // Capture an image from the camera (replace this with your camera capture code)
        let img: RgbImage = ImageBuffer::new(640, 480); // Dummy image

        // Convert the image to a ROS sensor_msgs/Image message
        let  ros_image = sensor_msgs::msg::Image {
            header: std_msgs::msg::Header {
                stamp: r2r::builtin_interfaces::msg::Time{sec: 1,nanosec: 1},
                ..Default::default()
            },
            height: img.height() as u32,
            width: img.width() as u32,
            encoding: String::from("rgb8"),
            is_bigendian: 0,
            step: (3 * img.width()) as u32,
            data: img.into_raw(),
        };

        // Publish the image
        publisher.publish(&ros_image).unwrap();

        // Sleep for the loop rate duration
        std::thread::sleep(loop_rate);
    }
}

closed 2023-12-27 by xiaocijun

m-dahl, 2023-12-27:
I assume you found `Clock::to_builtin_time` since you closed the issue? It's in a weird place, someday I will move it :)

xiaocijun, 2023-12-28:
> I assume you found `Clock::to_builtin_time` since you closed the issue? It's in a weird place, someday I will move it :) 

ok