sequenceplanner/r2r #81: Occasional build failures

istvan-fodor, 2024-01-11:
## Describe the bug
When I try to build the master branch (commit: cb87b9c01c3fee430a90fff352523e1fdd0511ac), I get sporadic build failures such as these: 

```
root@6a46f60839d9:/workspace/src/r2r# cargo build
    Updating crates.io index
   Compiling unicode-ident v1.0.12
....
   Compiling r2r_macros v0.8.2 (/workspace/src/r2r/r2r_macros)
error: failed to build archive: No such file or directory

error: could not compile `r2r_rcl` (lib) due to previous error
```

In the r2r_msg_gen folder: 

```
root@6a46f60839d9:/workspace/src/r2r/r2r_msg_gen# cargo build
   Compiling proc-macro2 v1.0.76
   ...
   Compiling r2r_common v0.8.2 (/workspace/src/r2r/r2r_common)
   Compiling syn v2.0.48
   Compiling rand_core v0.6.4
   Compiling rand v0.8.5
   Compiling phf_shared v0.11.2
   Compiling widestring v1.0.2
   Compiling phf_generator v0.11.2
   Compiling rayon v1.8.0
   Compiling itertools v0.10.5
   Compiling force-send-sync v1.1.0
   Compiling phf_macros v0.11.2
   Compiling r2r_rcl v0.8.2 (/workspace/src/r2r/r2r_rcl)
   Compiling phf v0.11.2
error: failed to build archive: No such file or directory

error: could not compile `r2r_rcl` (lib) due to previous error
```

Also in the r2r_msg_gen folder, but a different issue:

```
root@6a46f60839d9:/workspace/src/r2r/r2r_msg_gen# cargo build
   Compiling proc-macro2 v1.0.76
  ...
   Compiling r2r_rcl v0.8.2 (/workspace/src/r2r/r2r_rcl)
   Compiling phf v0.11.2
   Compiling r2r_msg_gen v0.8.2 (/workspace/src/r2r/r2r_msg_gen)
error: linking with `cc` failed: exit status: 1
  |
  = note: LC_ALL="C" PATH="/usr/local/rustup/toolchains/1.74.0-aarch64-unknown-linux-gnu/lib/rustlib/aarch64-unknown-linux-gnu/bin:/opt/ros/humble/bin:/usr/local/cargo/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin" VSLANG="1033" "cc" "/tmp/rustcceJls8/symbols.o" "/workspace/src/r2r/target/debug/build/r2r_msg_gen-e9ad2381dcda375b/build_script_build-e9ad2381dcda375b.12zdci3pgc82bt7m.rcgu.o" "/workspace/src/r2r/target/debug/build/r2r_msg_gen-e9ad2381dcda375b/build_script_build-e9ad2381dcda375b.13spihe4tc5d5ny1.rcgu.o" 
...
          /usr/bin/ld: cannot find /workspace/src/r2r/target/debug/build/r2r_msg_gen-e9ad2381dcda375b/build_script_build-e9ad2381dcda375b.x32cgeyuqc1jgto.rcgu.o: No such file or directory
          collect2: error: ld returned 1 exit status

error: could not compile `r2r_msg_gen` (build script) due to previous error
```

I trimmed down these errors a bit for readability, but I can provide the full errors if needed in the comment. 

Sometimes I can get a successful build, but even this is with some warnings here: 
```
root@6a46f60839d9:/workspace/src/r2r/r2r_msg_gen# cargo build
   Compiling typenum v1.17.0
   Compiling generic-array v0.14.7
   Compiling crypto-common v0.1.6
   Compiling block-buffer v0.10.4
   Compiling digest v0.10.7
   Compiling sha2 v0.10.8
   Compiling r2r_common v0.8.2 (/workspace/src/r2r/r2r_common)
   Compiling r2r_rcl v0.8.2 (/workspace/src/r2r/r2r_rcl)
   Compiling r2r_msg_gen v0.8.2 (/workspace/src/r2r/r2r_msg_gen)
error: failed to build archive: No such file or directory

error: could not compile `r2r_msg_gen` (lib) due to previous error

root@6a46f60839d9:/workspace/src/r2r/r2r_msg_gen# cargo build
   Compiling r2r_rcl v0.8.2 (/workspace/src/r2r/r2r_rcl)
   Compiling r2r_msg_gen v0.8.2 (/workspace/src/r2r/r2r_msg_gen)
warning: error copying object file `/workspace/src/r2r/target/debug/deps/r2r_msg_gen-00c58d0eab42599d.2620itydnjy74yt0.rcgu.o` to incremental directory as `/workspace/src/r2r/target/debug/incremental/r2r_msg_gen-175e4ubb88a6j/s-gsda2c1zr4-1ejczho-working/2620itydnjy74yt0.o`: No such file or directory (os error 2)

warning: `r2r_msg_gen` (lib) generated 1 warning
    Finished dev [unoptimized + debuginfo] target(s) in 9.73s
```
## To Reproduce
Clone master branch and run a build. 

## Expected behavior
Successful build. 

## Screenshots/Video
If applicable, add screenshots/video to help explain your problem.
Remember to mark the area in the application thats impacted.

## Build environment
 - Docker for Mac
 - devcontainer with `ghcr.io/devcontainers/features/rust:1` feature. 

## Additional context
I am building in a Docker container and it seems like I have less issues when I allocate fewer CPUs to my Docker

m-dahl, 2024-01-11:
Hi,

Is `/workspace/src/r2r` a bind mount? When I use bind mounts and virtiofs on mac I am also unable to build properly most of the time. If I copy the code into the container it works. Like this issue: https://github.com/rust-lang/docker-rust/issues/161#issue-1938684067

istvan-fodor, 2024-01-12:
Yes, thanks for the tip!

I verified that if I switch Docker for Mac to use gRPC FUSE it works without issues.

closed 2024-01-12 by istvan-fodor