sequenceplanner/r2r #82: Version 0.8.3 won't build, Error E0446

Druthyn, 2024-01-18:
Steps to reproduce:

- Make new directory
- run cargo init
- Add r2r = "0.8.3" to Cargo.toml
- run cargo build

Cargo build output below

```
cargo build
   Compiling r2r v0.8.3
error[E0446]: crate-private type  in public interface
   --> /home/druthyn/.cargo/registry/src/index.crates.io-6f17d22bba15001f/r2r-0.8.3/src/publishers.rs:128:1
    |
44  |   pub(crate) struct Publisher_ {
    |   ----------------------------  declared as crate-private
...
128 | / pub fn make_publisher<T>(handle: Weak<Publisher_>) -> Publisher<T>
129 | | where
130 | |     T: WrappedTypesupport,
    | |__________________________^ can't leak crate-private type

error[E0446]: crate-private type  in public interface
   --> /home/druthyn/.cargo/registry/src/index.crates.io-6f17d22bba15001f/r2r-0.8.3/src/publishers.rs:138:1
    |
44  | pub(crate) struct Publisher_ {
    | ----------------------------  declared as crate-private
...
138 | pub fn make_publisher_untyped(handle: Weak<Publisher_>, type_: String) -> PublisherUntyped {
    | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ can't leak crate-private type

error[E0446]: crate-private type  in public interface
   --> /home/druthyn/.cargo/registry/src/index.crates.io-6f17d22bba15001f/r2r-0.8.3/src/publishers.rs:142:1
    |
44  |   pub(crate) struct Publisher_ {
    |   ----------------------------  declared as crate-private
...
142 | / pub fn create_publisher_helper(
143 | |     node: &mut rcl_node_t, topic: &str, typesupport: *const rosidl_message_type_support_t,
144 | |     qos_profile: QosProfile,
145 | | ) -> Result<Publisher_> {
    | |_______________________^ can't leak crate-private type

For more information about this error, try 'rustc --explain E0446'
error: could not compile  (lib) due to 3 previous errors
```

Version 0.8.2 builds without issue for me.

m-dahl, 2024-01-18:
Hi, thanks for the notice. I cannot reproduce locally and it seems to build fine in CI (see https://github.com/sequenceplanner/r2r/actions/runs/7518385987/job/20607569406#step:5:10)
Which rust version are you running?

Druthyn, 2024-01-18:
I was on 1.73. I can confirm updating to 1.75 has resolved this problem.

My bad for not checking, thanks for the quick response :)

closed 2024-01-18 by Druthyn