sequenceplanner/r2r #83: BUG: Canno't compile library

JeromeCGithub, 2024-01-28:
Hello,
I'm trying to use your library, i got this error : 
Compiling r2r v0.8.3
error[E0446]: crate-private type `publishers::Publisher_` in public interface
   --> /*/r2r-0.8.3/src/publishers.rs:128:1
    |
44  |   pub(crate) struct Publisher_ {
    |   ---------------------------- `publishers::Publisher_` declared as crate-private
...
128 | / pub fn make_publisher<T>(handle: Weak<Publisher_>) -> Publisher<T>
129 | | where
130 | |     T: WrappedTypesupport,
    | |__________________________^ can't leak crate-private type

Any idea why i canno't compile it ?

I'm using this version : r2r = "0.8.3"

NerdToMars, 2024-01-29:
A bit hard for me to tell what causes this error, 

Maybe you can share the steps to reproduce this bug?

m-dahl, 2024-01-29:
Hi, looks like #82. Try to update your rust version.