sequenceplanner/r2r #93: Question about the `spin_once`

A-K-O-R-A, 2024-04-24:
How does the `spin_once` function differ from it's python equivalent? Looking at the [rclpy](https://github.com/ros2/rclpy/tree/rolling) source code, the [`spin`](https://github.com/ros2/rclpy/blob/e181737f19d6b8cb335b0326a40ec2075267c555/rclpy/rclpy/executors.py#L305) it seems like it is also just calling `spin_once` in a loop. But the big difference is that the python code for a minimal node that does nothing, takes about 4% of CPU usage, while the `r2r` Node instantly utilizes 100% of one core.

Rust code:
```rs
pub fn main() -> Result<(), Box<dyn std::error::Error>> {
    let ctx = r2r::Context::create()?;
    let mut node = r2r::Node::create(ctx, "test", "test")?;

    loop {
        node.spin_once(std::time::Duration::from_millis(500));
    }
}

```

Python code:
```py
import rclpy

rclpy.init()
test_node = rclpy.create_node('test_node')
while rclpy.ok():
  rclpy.spin_once(test_node)
```

Is there a way to achieve a behavior similar to the python one? ( For my use case it's necessary that the CPU usage stays relatively low )

m-dahl, 2024-04-24:
Hi.

I think what is happening here is that `rcl_wait` returns early because there is nothing in the wait set. If you just add for example a subscriber (even without doing anything with it) I think you will get similar numbers as with rclpy.

I don't know if python sleeps a little anyway if it detects that the wait set is empty. Doing that would probably solve this problem but I am not sure it is a problem in practice since you are always waiting on something (or else you would not need to spin).

A-K-O-R-A, 2024-04-24:
Thank you :) this did in fact solve the issue. If you don't mind I do have another question though. Running the following example uses about 2% of the CPU, simply uncommenting the creation of the `_sub2` part, makes the CPU usage go to 100%. I struggle to come up with an explanation for this and was hoping you could help me

```rs
use r2r::QosProfile;

pub fn main() -> Result<(), Box<dyn std::error::Error>> {
    let ctx = r2r::Context::create()?;
    let mut node = r2r::Node::create(ctx, "test", "test")?;

    for _ in 0..50 {
        let _sub1 = node.subscribe::<r2r::std_msgs::msg::String>(
            "/this/gets_published_every_5ms",
            QosProfile::default(),
        )?;
    }

    let _sub2 = node.subscribe::<r2r::std_msgs::msg::String>(
         "/this/wont/get/published",
         QosProfile::default(),
    )?;

    loop {
        node.spin_once(std::time::Duration::from_millis(5));
    }
}

```
( The published string is just "A" )

m-dahl, 2024-04-24:
Hi again. It's because your `_sub1`:s get dropped by at the end of the scope of the for loop. So then you are back to nothing to wait for when you comment out `_sub2`. If you were to put the `_sub1` in a vec to keep them alive you would not get the 100% cpu use. I hope this helps.

A-K-O-R-A, 2024-04-25:
Thank you for the explanation, this helped a lot, I feel a bit stupid now :)
Btw I love this project and it was exactly what I was looking for so thank you for making this <3

m-dahl, 2024-04-26:
Thanks for the kind words!

closed 2024-04-26 by m-dahl