sequenceplanner/r2r #103: Support ESP32

Ryosuke1218, 2024-09-07:
Hi, I'm struggling compiling this crate(version 0.9.0) with ESP32. 
I'm new to Rust and could not determine if it is still possible in its current state.
If you haven't already, it would be very helpful to have an environment where r2r can be used with ESP32.

error[E0433]: failed to resolve: could not find `Access` in `rfs`
  --> /home/ryou/.cargo/registry/src/index.crates.io-6f17d22bba15001f/which-4.4.2/src/checker.rs:17:32
   |
17 |         rfs::access(path, rfs::Access::EXEC_OK).is_ok()
   |                                ^^^^^^ could not find `Access` in `rfs`

error[E0425]: cannot find function `access` in module `rfs`
  --> /home/ryou/.cargo/registry/src/index.crates.io-6f17d22bba15001f/which-4.4.2/src/checker.rs:17:14
   |
17 |         rfs::access(path, rfs::Access::EXEC_OK).is_ok()
   |              ^^^^^^ not found in `rfs`

Some errors have detailed explanations: E0425, E0433.
For more information about an error, try `rustc --explain E0425`.
error: could not compile `which` (lib) due to 2 previous errors
warning: build failed, waiting for other jobs to finish...
error[E0425]: cannot find value `X_OK` in crate `libc`
   --> /home/ryou/.cargo/registry/src/index.crates.io-6f17d22bba15001f/clang-sys-1.8.1/src/support.rs:165:51
    |
165 |     unsafe { Ok(libc::access(path.as_ptr(), libc::X_OK) == 0) }
    |                                                   ^^^^ not found in `libc`

error: failed to run custom build command for `r2r_actions v0.9.0`

Caused by:
  process didn't exit successfully: `/home/ryou/Documents/r2r-test/target/debug/build/r2r_actions-4f9d2b344ad7888a/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=AMENT_PREFIX_PATH
  cargo:rerun-if-env-changed=CMAKE_PREFIX_PATH
  cargo:rerun-if-env-changed=CMAKE_IDL_PACKAGES
  cargo:rerun-if-env-changed=IDL_PACKAGE_FILTER
  cargo:rerun-if-env-changed=ROS_DISTRO
  cargo:rustc-cfg=r2r__ros__distro__humble

  --- stderr
  Generate bindings file '/home/ryou/Documents/r2r-test/target/xtensa-esp32-espidf/debug/build/r2r_actions-cbb5d70140a9f74d/out/a0d63c11f6aab0746704a41f6145a7033ff3fb816a6b30e5a5acf48d70935a1a/action_bindings.rs'
  error: unknown target triple 'xtensa-esp32-espidf', please use -triple or -arch
  thread 'main' panicked at /home/ryou/.cargo/registry/src/index.crates.io-6f17d22bba15001f/bindgen-0.63.0/ir/context.rs:530:15:
  libclang error; possible causes include:
  - Invalid flag syntax
  - Unrecognized flags
  - Invalid flag arguments
  - File I/O errors
  - Host vs. target architecture mismatch
  If you encounter an error missing from this list, please file an issue or a PR!
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
error: failed to run custom build command for `r2r_rcl v0.9.0`

Caused by:
  process didn't exit successfully: `/home/ryou/Documents/r2r-test/target/debug/build/r2r_rcl-48fcbd6ba27cfb8a/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=AMENT_PREFIX_PATH
  cargo:rerun-if-env-changed=CMAKE_PREFIX_PATH
  cargo:rerun-if-env-changed=CMAKE_IDL_PACKAGES
  cargo:rerun-if-env-changed=IDL_PACKAGE_FILTER
  cargo:rerun-if-env-changed=ROS_DISTRO
  cargo:rustc-cfg=r2r__ros__distro__humble

  --- stderr
  Generate bindings file '/home/ryou/Documents/r2r-test/target/xtensa-esp32-espidf/debug/build/r2r_rcl-746b399480a015bd/out/a0d63c11f6aab0746704a41f6145a7033ff3fb816a6b30e5a5acf48d70935a1a/rcl_bindings.rs'
  error: unknown target triple 'xtensa-esp32-espidf', please use -triple or -arch
  thread 'main' panicked at /home/ryou/.cargo/registry/src/index.crates.io-6f17d22bba15001f/bindgen-0.63.0/ir/context.rs:530:15:
  libclang error; possible causes include:
  - Invalid flag syntax
  - Unrecognized flags
  - Invalid flag arguments
  - File I/O errors
  - Host vs. target architecture mismatch
  If you encounter an error missing from this list, please file an issue or a PR!
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
error: failed to run custom build command for `r2r_msg_gen v0.9.0`

Caused by:
  process didn't exit successfully: `/home/ryou/Documents/r2r-test/target/debug/build/r2r_msg_gen-ab6e3a62d4fa7cf4/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=AMENT_PREFIX_PATH
  cargo:rerun-if-env-changed=CMAKE_PREFIX_PATH
  cargo:rerun-if-env-changed=CMAKE_IDL_PACKAGES
  cargo:rerun-if-env-changed=IDL_PACKAGE_FILTER
  cargo:rerun-if-env-changed=ROS_DISTRO
  cargo:rustc-cfg=r2r__ros__distro__humble

  --- stderr
  Generate bindings in '/home/ryou/Documents/r2r-test/target/xtensa-esp32-espidf/debug/build/r2r_msg_gen-5d2130f729d84370/out/a0d63c11f6aab0746704a41f6145a7033ff3fb816a6b30e5a5acf48d70935a1a'
  error: unknown target triple 'xtensa-esp32-espidf', please use -triple or -arch
  thread '<unnamed>' panicked at /home/ryou/.cargo/registry/src/index.crates.io-6f17d22bba15001f/bindgen-0.63.0/ir/context.rs:530:15:
  libclang error; possible causes include:
  - Invalid flag syntax
  - Unrecognized flags
  - Invalid flag arguments
  - File I/O errors
  - Host vs. target architecture mismatch
  If you encounter an error missing from this list, please file an issue or a PR!
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
For more information about this error, try `rustc --explain E0425`.
error: could not compile `clang-sys` (lib) due to 1 previous error

m-dahl, 2024-10-01:
Hi,

Sorry I don't think it is possible, since we link against the ROS version installed on your computer. So we need shared libraries, we use the file system etc.

m-dahl, 2024-11-21:
Closing due to inactivity

closed 2024-11-21 by m-dahl