sequenceplanner/r2r #108: Checking of cfgs at compile-time.

m-dahl, 2024-10-12:
Since rustc 1.80 we get a lot of warnings of the type: `warning: unexpected `cfg` condition name`. I need to read up on the change and see how to best align with how rust/cargo wants to do it in newer rust versions.

closed 2024-10-13 by m-dahl

jdiez17, 2024-11-15:
Hi, this is causing build errors on our cross compilation setup. We are targeting Rust 1.75 (the default in Yocto scarthgap). We get the following build error:

```
error: unsupported output in build script of `r2r v0.9.2`: `cargo::rustc-check-cfg=cfg(r2r__ros__distro__foxy)`
Found a `cargo::key=value` build directive which is reserved for future use.
Either change the directive to `cargo:key=value` syntax (note the single `:`) or upgrade your version of Rust.
```

Reverting to r2r 0.9.1 (without `cargo::rustc-check`) fixes it. Is it possible to make a backwards-compatible fix?

m-dahl, 2024-11-19:
Hi,

Thanks for noticing. I pushed a fix. You will get warnings though but I would rather not have conditional code based on the rust version if we can avoid it.

commit 4cb5bd36 (2024-10-13): Silence cfg and ffi warnings. Fixes #108.