sequenceplanner/r2r #109: Impossible to destroy a publisher

showier-drastic, 2024-10-21:
Hi,

It seems that we cannot destroy a publisher created from the node, because the `destroy` function is private to the inner `Publisher_` type. The only way to remove a publisher is to close the node.

Is it possible to add `Drop` to `Publisher<T>` so we can drop it when necessary?

m-dahl, 2024-10-30:
Hi,

You are right this functionality is missing. If I remember correctly it is like this is because the node should be the one destroying the publishers so that they can be shared between threads easily. But it seems that there is no function on the node for destroying a publisher either which there ought to be.

m-dahl, 2024-10-30:
Would the solution in #110 work for you? It does not implement drop but it was the simplest way to add the missing functionality.

showier-drastic, 2024-10-31:
Thanks for implementing this! Looks good to me.

closed 2024-11-02 by m-dahl