sequenceplanner/r2r #116: Compilation trouble on Ubuntu 24 with ROS2 Iron

gjf2a, 2025-02-19:
Hello, I am creating a project that runs perfectly well on Ubuntu 22 with ROS2 Iron. 

Unfortunately, I also need it to run on Ubuntu 24 with ROS2 Iron, which required me to build ROS2 Iron from sources. That generally works well - I can control my robot just fine. What I can't do is make my Rust code work with it. 

Here is my repository: https://github.com/gjf2a/trajectory_mapper

When I try to compile it on Ubuntu 22, everything works fine. When I try on Ubuntu 24, I get the following error:

```
  Generate bindings file '/home/ferrer/trajectory_mapper/target/debug/build/r2r_rcl-0f95862c1d61477e/out/f2001a1aae781924ad5d5e8f16a78e1e08e71423ac05f9a98f06392f545f4268/rcl_bindings.rs'
  src/rcl_wrapper.h:27:10: fatal error: 'rosidl_runtime_c/rosidl_runtime_c.h' file not found
  thread 'main' panicked at /home/ferrer/.cargo/registry/src/index.crates.io-6f17d22bba15001f/r2r_rcl-0.9.4/build.rs:103:10:
  Unable to generate bindings: ClangDiagnostic("src/rcl_wrapper.h:27:10: fatal error: 'rosidl_runtime_c/rosidl_runtime_c.h' file not found\n")
```

As an experiment, I used `touch` to create a bogus `rosidl_runtime_c.h`, and I got this error:

```
error[E0412]: cannot find type `rosidl_runtime_c__U16String__Sequence` in this scope
    --> /home/ferrer/.cargo/registry/src/index.crates.io-6f17d22bba15001f/r2r_rcl-0.9.4/src/lib.rs:55:6
     |
55   | impl rosidl_runtime_c__U16String__Sequence {
     |      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ help: a type alias with a similar name exists: `rosidl_runtime_c__byte__Sequence`
     |
    ::: /home/ferrer/trajectory_mapper/target/debug/build/r2r_rcl-0f95862c1d61477e/out/rcl_bindings.rs:1713:1
     |
1713 | pub type rosidl_runtime_c__byte__Sequence = [u64; 3usize];
     | ---------------------------------------------------------- similarly named type alias `rosidl_runtime_c__byte__Sequence` defined here

```

I would greatly appreciate any and all advice you can offer. Thanks!

wentasah, 2025-03-19:
I'm seeing the **cannot find type `rosidl_runtime_c__U16String__Sequence` in this scope** error too. But with Jazzy and when building with the Nix package manager. For some reason, `bindgen` doesn't generate definition for `rosidl_runtime_c__U16String__Sequence` in `rcl_bindings.rs`. This is defined in `rosidl` [here](https://github.com/ros2/rosidl/blob/1d92ce395ca9afd9cba59a4af6323218b7025600/rosidl_runtime_c/include/rosidl_runtime_c/u16string.h#L33). Interestingly, functions like `rosidl_runtime_c__U16String__Sequence__init`, are generated.

So far, I've not figured out what is the reason. When I was using older version of Nix packages, this worked well. I'll continue with the investigation later.

gjf2a, 2025-03-19:
Thanks @wentasah! For now, I do have a provisional solution - I am compiling on a Raspberry Pi 4 board running Ubuntu 22, and then copying the compiled executable onto the Raspberry Pi 5 boards. I had to copy over some shared object files to make everything work, but for now it is a workable solution. I would definitely be interested in a more long-term solution, though!

beuss-git, 2025-03-26:
> So far, I've not figured out what is the reason. When I was using older version of Nix packages, this worked well. I'll continue with the investigation later.

Hey @wentasah , I am not sure if you got any further into the investigation, but updating bindgen fixed the issue for me.
I submitted the fix in #118.

commit 56acba5d (2025-04-17): Update bindgen 0.63.0 -> 0.71.1 (#116)

This fixes #116 by now generating the missing primitive sequence functions.
Tested to be working on Ubuntu 22.04 with ROS2 Humble and Iron.
closed 2025-04-17 by m-dahl