sequenceplanner/r2r #119: Issues with parameter services

des256, 2025-04-10:
I find that on ROS2 Jazzy, the minimal example `parameters.rs` doesn't work. I can't access the parameters externally via `ros2 param ...`. We need parameters in our project, so I'm going to debug this, but I'm curious if anyone else has any problems with this.

I'm currently trying:

- The service example in `r2r_minimal_node` (`https://github.com/m-dahl/r2r_minimal_node`) works fine for `ros2 service ...`. It doesn't respond to `ros2 param ...` as expected (it doesn't start a parameter handler). When I add the parameter handler from the `parameters` example, it still doesn't respond to `ros2 param ...`.

- With debug statements inside the parameter handler code (`r2r/src/parameters.rs`), it looks like even the requests are not coming in. So it looks like a lower-level communication issue.

I'll keep you posted when I find anything...

des256, 2025-04-10:
This does work: `ros2 service call /demo/my_node/list_parameters rcl_interfaces/srv/ListParameters`

But this doesn't: `ros2 param list`

wentasah, 2025-04-10:
Confirming that `ros2 param list` doesn't work. I set parameters with `rig_reconfigure` and Foxglove and these work well. I tried `rqt` and this also has problems with r2r nodes.

des256, 2025-04-10:
On humble it works. But humble requires Ubuntu 22. Contemplating if we can migrate the whole project...

m-dahl, 2025-04-22:
Closed by #120, included in 0.9.5 release. Thanks!

closed 2025-04-22 by m-dahl